cawilliamson/treble_voltage #1: Bluetooth crashes/doesn't work

tommynok, 2024-01-16:
Device -Tecno Camon 19Neo
OS 12A
MTK Helio G85
[bluetooth.zip](https://github.com/cawilliamson/treble_voltage/files/13952735/bluetooth.zip)

tommynok, 2024-01-16:
> Does Bluetooth work properly on other TrebleDroid based GSI ROMs? If so, which?

Yes. 
Latest AOSP, TD

cawilliamson, 2024-01-17:
Ok folks - can you try the folllowing *WITHOUT* rebooting?

```
adb shell
setprop bluetooth.hci.disabled_commands $(getprop persist.sys.bt.unsupported.commands)
```

Then try your bluetooth pairing again? It MAY need BT turning off and on again.

If that works let me know and I'll get a patch merged in.

cawilliamson, 2024-01-17:
Please try this build and report back:

https://github.com/cawilliamson/treble_voltage/releases/tag/3.1-20240117-2

Either way - could you post back the output of the following command?

```
adb shell getprop bluetooth.hci.disabled_cpmmands
```

tommynok, 2024-01-17:
> Should solve the issue once and for all!
A quick test of the new build -
after installation, a bluetooth error appears immediately on the initial installation screen (bluetooth has stopped....)
As in previous builds
But bluetooth works ok.
Of course I had to turn on "bluetooth workarounds-mediatek" in phh settings

cawilliamson, 2024-01-17:
Does it do that every boot?

Glad to hear things are working properly other than that though. That's certainly progress. :)

tommynok, 2024-01-17:
> Does it do that every boot?
You mean "bluetooth has stopped...." error ? If yes, then no.
This appears only at the first start, initial system setup
Thanks for the fixes

closed 2024-01-17 by tommynok