cawilliamson/treble_voltage #2: Bluetooth crashes on medietake devices

walker425, 2024-01-17:
Bluetooth doesn't work on medietake devices even when I turn on work around from phh settings. It still crashes and another thing is the clock styles themes no longer work in this new version 3.1 but 3.0 everything is working well except this very new update 3.1.

cawilliamson, 2024-01-17:
I'm on a MediaTek device - works fine for me. Also this is a duplicate of #1 so closing in favour of that.

Please add more information to that ticket about your device, vendor version, etc.

Also clock styles will be an issue with Voltage itself - not my build.

closed 2024-01-17 by cawilliamson

walker425, 2024-01-17:
Ok my device is infinix note 11 and shipped with Android 11 out of the box but I upgraded to Android version 12 before flashing a gsi ROM , so I think I am on Android 12 vendor.

walker425, 2024-01-17:
Honestly I was on voltage os 3.0 gsi by you and everything was working even Bluetooth was working flawlessly but this new build is where I have a problem with Bluetooth.