cawilliamson/treble_voltage #3: why only build VNDK bvN ??

LeOS-GSI, 2024-01-17:
Why don't build also VNDKLITE bvN and bvS and a64 versions ??

cawilliamson, 2024-01-17:
I have zero interest in building bvS because it's THE WORST superuser option out there - detectable by everything, breaks ALL banking apps, breaks ALL work apps which are checking for root. Just a terrible idea.

Use either:
- APatch
- KernelSU
- Magisk (or one of the many many forks)

Nobody has asked for a64 and nobody has (yet) asked for vndklite. Building takes a lot of computational time as I'm sure you're aware being a GSI maintainer (I'd guess based on your name) yourself. If I do 5 builds instead of 1 each time - that delays fixes massively.

LeOS-GSI, 2024-01-17:
I'm always building every month all 6 versions for A13 and A14 and my community is happy about that. I'm reading often in PHH group that people want have a VNDKLITE version and or bvS. That's why I have asked.

cawilliamson, 2024-01-17:
Ah understood.

I may build a vndklite build but bvS isn't coming since as I mentioned - it's just a terrible way to root your device and vastly better options exist.

cawilliamson, 2024-01-17:
So the conclusion on this:

- I don't build bvS and don't intend to because there are much better options.
- I don't build a64 because there are well known build issues with Voltage on a64 and I've never had any requests for a64 at all so unless a dependable tester appears - I'll leave that out.
- I will look in to building vndklite builds going forward but again - I've never had anyone ask for these on the channel so not too worried about it yet.

With the above in mind I'll close the ticket - thanks for the idea about vndklite though - will pursue that.

closed 2024-01-17 by cawilliamson