cawilliamson/treble_voltage #6: [question] patches for storage scopes

LeOS-GSI, 2024-01-23:
Hi, 
hhave you patches for the storage scopes features. I can't find any for packages/apps/Settings.
Vor is it hard coded in you packages/apps/Settings repo ?

cawilliamson, 2024-01-26:
We talked about this on the TrebleDroid telegram channel - to repeat here for anyone else though: I don't know exactly where the patches are applied since this is done by the VoltageOS team themselves. Not me!

Best of luck digging them out though. :)

closed 2024-01-26 by cawilliamson

LeOS-GSI, 2024-01-26:
I have used a different way. I have build a LeOS-GSI based on your voltage sources 😁. It's in testing phase now. I have removed most default apps, added my LeOS stuff. 
In the moment I'm searching for a way to stop the force encryption. But all other is working find. Nice features, good work 👍