cawilliamson/treble_voltage #7: VoltageOS 3.1-20240124 and Samsung A21s. UI Gestures and power button not working anymore

fparri, 2024-01-30:
I have just tested VoltageOS 3.1-20240124. Unlike the previous release I am using (VoltageOS 3.1-20240121-2), which is working fine for me, the latest one is not compatible with my Samsung A21s. Specifically, I cannot control the UI (no gestures) and the power button does not function (to turn on/off the screen or to shut down the phone).

Do you have any suggestions on how I can help to solve this issue? :)

cawilliamson, 2024-01-30:
That's highly strange - I don't see any obvious changes which would affect gestures, etc.

Could you try the new build I just uploaded today and see if that's any different? :)

I can only assume it was a ROM-side change and perhaps now it's resolved at their end! :+1:

fparri, 2024-01-31:
I've tried the latest release and everything is now working perfectly. Thank you for your assistance and for the new build. It seems like any issues were resolved on the ROM side. 👍

closed 2024-01-31 by fparri