cawilliamson/treble_voltage #10: Disable booting for Samsung S23 QCOM Series.

Edgars-Cirulis, 2024-02-01:
Hello there. Can you disable Samsung S23 QCOM series booting? Would really Appreciate. 

WiWk, That changelog "Enable Galaxy S24 booting"  is nonsense.

cawilliamson, 2024-02-04:
What? I don't understand what this means...

Why would you want a device to NOT boot anymore?

closed 2024-02-05 by cawilliamson

Edgars-Cirulis, 2024-02-05:
> Closing this one since it makes no sense. Please re-open if you can clarify what you actually want here.

"Enables Booting for S24" May you please describe the non-sense in which commit it enables?

cawilliamson, 2024-02-06:
No. Next time show some respect.

Edgars-Cirulis, 2024-02-06:
> No. Next time show some respect.

Then don't Write fake and nonsense changelog.

Edgars-Cirulis, 2024-02-06:
Let me get you straight Into it, 
The patch in `AuthService.java`, it adds a static method `dynamicUdfpsProps(Context ctxt)` which retrieves dynamic UDFPS (Under Display Fingerprint Sensor) properties using the `DisplayManager`. It's made static and public so it can be accessed from other classes.
WiWi, 
In `FingerprintProvider.java`, it overrides UDFPS (Under Display Fingerprint Sensor) properties on AIDL (Android Interface Definition Language) fingerprints. It checks if the sensor location is at (0, 0) and if so, it replaces the location with dynamically retrieved UDFPS properties using the `AuthService.dynamicUdfpsProps(context)` method.

I do not see anything that's related to boot stuff.

cawilliamson, 2024-02-06:
Problem solved.