cawilliamson/treble_voltage #9: Request for a64a BNV VNDKlite GSI

Abeefe, 2024-02-01:
I love your GSI though i can onlu used LeOS fork of it which is degoogle and i have that i help install GSI on their phone but they use devices that is a6a version.

I will love they to use your GSI since it has lots of privacy enhancement.

cawilliamson, 2024-02-05:
I'll look in to this!

Abeefe, 2024-02-05:
OK. Thanks.

cawilliamson, 2024-02-06:
I have now added a build for a64 (aka arm32_binder64) here:

https://github.com/cawilliamson/treble_voltage/releases/tag/3.1-20240206

Please open a new ticket if you find any problems. I suspect face unlock MAY not work - not 100% sure.

closed 2024-02-06 by cawilliamson

Abeefe, 2024-02-08:
Thanks for your effort. I have been testing it since yesterday, seems no issues.
No face unlock in security settings to test it but i feel i don't need it.

Thanks again.