cawilliamson/treble_voltage #12: .img - corrupted file

timur0409, 2024-02-09:
[VoltageOS-arm64-ab-vndklite-3.2-20240206-UNOFFICIAL.img.xz](https://github.com/cawilliamson/treble_voltage/releases/download/3.2-20240206/VoltageOS-arm64-ab-vndklite-3.2-20240206-UNOFFICIAL.img.xz)
.img - corrupted file

cawilliamson, 2024-02-09:
I've just re-downloaded the image and verified at my end - no issue:

```
$ xz -d VoltageOS-arm64-ab-vndklite-3.2-20240206-UNOFFICIAL.img.xz                                                                            $ fsck.ext4 -f VoltageOS-arm64-ab-vndklite-3.2-20240206-UNOFFICIAL.img
e2fsck 1.47.0 (5-Feb-2023)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
/: 8659/9408 files (8.2% non-contiguous), 669622/670549 blocks
```

Please try re-downloading. :)

closed 2024-02-09 by cawilliamson

timur0409, 2024-02-10:
[image]

 re-downloading x4 img vndklite corrupted