cawilliamson/treble_voltage #13: Build Error

kelexine, 2024-02-17:
I got this error while building:
`FAILED: /tmp/src/android/voltage/volt/out/soong/bp2build_files_marker
cd "$(dirname "/tmp/src/android/voltage/volt/out/host/linux-x86/bin/soong_build")" && BUILDER="$PWD/$(basename "/tmp/src/android/voltage/volt/out/host/linux-x86/bin/soong_build")" && cd / && env -i  "$BUILDER"     --top "$TOP"     --soong_out "/tmp/src/android/voltage/volt/out/soong"     --out "/tmp/src/android/voltage/volt/out"     --bp2build_marker /tmp/src/android/voltage/volt/out/soong/bp2build_files_marker --globListDir bp2build_files --globFile /tmp/src/android/voltage/volt/out/soong/globs-bp2build_files.ninja -t -l /tmp/src/android/voltage/volt/out/.module_paths/Android.bp.list --available_env /tmp/src/android/voltage/volt/out/soong/soong.environment.available --used_env /tmp/src/android/voltage/volt/out/soong/soong.environment.used.bp2build_files Android.bp
error: hardware/qcom-caf/sm8550/display/sdm/libs/core/Android.bp:1:1: "libsdmcore" depends on undefined module "sdmcore_has_is_display_hw_available_func_defaults".
Or did you mean ["core-gen-test-repackaged-java-11-language-features" "core-gen-test-repackaged-java-17-language-features" "core-gen-test-repackaged-java-9-language-features" "core-java-11-language-features-repackaged-for-test" "core-java-17-language-features-repackaged-for-test" "core-java-9-language-features-repackaged-for-test" "core-rewrite-java-11-test-for-jarjar-sed-script" "core-rewrite-java-17-test-for-jarjar-sed-script"]?
00:27:09 soong bootstrap failed with: exit status 1

#### failed to build some targets (24 seconds) ####
`

cawilliamson, 2024-02-17:
Yeah seeing the same issue at my end. Strongly believe this is an upstream breakage since there's been no major changes at my end to cause this.

Btw - are you planning to fork my GSI work? No problem if so - would be interesting to see Voltage go in a different direction.

kelexine, 2024-02-17:
Yeah, already forked it

kelexine, 2024-02-17:
> Yeah seeing the same issue at my end. Strongly believe this is an upstream breakage since there's been no major changes at my end to cause this.
> 
> Btw - are you planning to fork my GSI work? No problem if so - would be interesting to see Voltage go in a different direction.

Okay, if you find any fix let me know

cawilliamson, 2024-02-17:
https://github.com/cawilliamson/treble_voltage/blob/54176e92fdc1bcb7840a08675f77b9cc3747f336/patches/personal/platform_vendor_voltage/0007-Add-missing-sdmcore_has_is_display_hw_available_func.patch

Yep - this is an issue with the voltage vendor but this patch appears to resolve the issue. I suspect this will be fixed upstream soon though so the patch will likely not be needed for long.

cawilliamson, 2024-02-17:
Re-opened - this patch fixes the initial issue but build is still failing. Investigating...

cawilliamson, 2024-02-18:
Now the webview build issue is solved upstream.

One of my patches is now failing the build but that's an easy fix - will merge it shortly.

cawilliamson, 2024-02-18:
Patch fix merged - will close this once I get a successful build. :)

kelexine, 2024-02-18:
> Now the webview build issue is solved upstream.
> 
> One of my patches is now failing the build but that's an easy fix - will merge it shortly.

Okay, will sync up with your repo and start a build too

cawilliamson, 2024-02-18:
Yep, build issue solved now! Closing ticket. :)

closed 2024-02-18 by cawilliamson

kelexine, 2024-02-18:
Yeah, just saw it now, my build is now working