cawilliamson/treble_voltage #14: Restart app game while change Black Mode

roachsoft, 2024-02-21:
Hi @cawilliamson!

Here my log from issue.

Thank you for all wonderfull Woks!!! :100: 

My device is Samsung S23 Ultra

[logCatFreeFire.txt](https://github.com/cawilliamson/treble_voltage/files/14364147/logCatFreeFire.txt)

cawilliamson, 2024-02-22:
Can you describe exactly what the issue is? Or provide a small video of the problem?

I don't understand what you mean in the issue title.

Thanks for the logs though.

roachsoft, 2024-02-22:
Hi!
I attach vídeo!

Thank you very much!!!

https://github.com/cawilliamson/treble_voltage/assets/7821308/ba6a4a07-1add-47bf-bfce-0a0e69fc89f8

cawilliamson, 2024-02-22:
How did you install the game? Sandboxed Gapps?

Looking through your logs it looks like this is likely the issue:

```
02-21 15:58:46.315  8635  8965 E AndroidRuntime: FATAL EXCEPTION: [com.google.android.gms.chimera.container.intentoperation.GmsIntentOperationChimeraService-Executor] idle
02-21 15:58:46.315  8635  8965 E AndroidRuntime: Process: com.google.android.gms, PID: 8635
02-21 15:58:46.315  8635  8965 E AndroidRuntime: java.lang.SecurityException: Attempt to change component state; pid=8635, uid=10204, package=com.google.android.gms.supervision
```

I've never seen that error before on any of my test devices while running sandboxed gapps so it seems very strange!

roachsoft, 2024-02-23:
I install from Google Playstore.
Then to try another alternative, uninstall this one, and try installing from apk mirror and the same thing happened.

Thank you for tour help!!!

cawilliamson, 2024-02-29:
I suspect this issue is with either the game or VoltageOS itself I'm afraid!

There's not much I can do to try to fix this in my GSI. Sorry!

closed 2024-02-29 by cawilliamson