cawilliamson/treble_voltage #20: im getting touch delay on the latest version and ui is quite laggy

Divyansh-2903, 2024-03-09:
its why, is this the problem with specific devices ? like in newer version of evolutionX have same problem but the last version of evox from koysX doesnt have that kind of problem , can you provide me a fix for this?

talksickisil, 2024-03-10:
I have issues with both voltage launcher and trebuchet on this last update.Things like crashing when I try to change to custom icons or change grid si,ze etc. So I just use SMRT launcher instead. I use a galaxy tab A and for A14, only this and lineage fit my system partition. This is better than lineage by far.

cawilliamson, 2024-03-10:
> its why, is this the problem with specific devices ? like in newer version of evolutionX have same problem but the last version of evox from koysX doesnt have that kind of problem , can you provide me a fix for this?

That sounds quite strange since the only real change in the latest release is Trebuchet which others have reported is working fantastically. Have you tried tweaking the touch boost related options in TrebleApp?

Could you giev me some more info about your specific device?

> I have issues with both voltage launcher and trebuchet on this last update.Things like crashing when I try to change to custom icons or change grid si,ze etc. So I just use SMRT launcher instead. I [...]

Your issue sounds completely unrelated - you should raise another issue.

cawilliamson, 2024-03-14:
Without additional info being provided there isn't much I can do to investigate this so closing for now.

Divyansh-2903, 2024-03-22:
No I'm getting touch response issue in almost all Android 14 roms , but there are some roms which are working fine for eg evolutionX by koysX ver 8.0.3

cawilliamson, 2024-05-17:
I suggest you raise this at https://github.com/TrebleDroid/treble_experimentations/issues since it seems like a device specific issue. Clearly some patch difference between KoysX builds and mine are causing the issue.

closed 2024-05-17 by cawilliamson