cawilliamson/treble_voltage #23: Stuck at boot animation

Jazkieplayzz000, 2024-03-15:
I've been stuck at the boot animation for 1 hour and still my phone did not go through the system btw im using mediatek device

cawilliamson, 2024-03-15:
I use a MediaTek device and have no such issue.

Can you provide more information and preferably logs if possible?

Did you clean flash? Does this happen with other ROMs? etc.

Jazkieplayzz000, 2024-03-15:
No, just only this latest rom. But i cannot send the logs due to stuck at
boot animations

cawilliamson, 2024-03-18:
Nobody else has reported any bootloops on this ROM - I suspect this may be a flashing issue.

Try downloading fresh and doing a clean flash on your device (i.e. wipe data, etc.)

Jazkieplayzz000, 2024-03-19:
I tried clean flashing but no hope on latest Release of your os still stuck
in the boot for almost 1 hour by the way may device is realme C11 (rmx2185
) but it works on the build VoltageOS 3.2-20240308

shunceyb, 2024-03-22:
I'm also having the same issue on my Xiaomi 13T(Dimensity 8200 Ultra)

cawilliamson, 2024-04-03:
Try the new build once it comes out and re-open the issue if still present. :)

It's based on aosp-14-r31 so it's WAY more up-to-date.

closed 2024-04-03 by cawilliamson