cawilliamson/treble_voltage #24: No Micro G

Abeefe, 2024-03-17:
I flashed the recent build: VoltageOS-microg-arm32_binder64-ab-vndklite-3.2-20240316-UNOFFICIAL

There is no micro g, I search the whole system.

cawilliamson, 2024-03-17:
Yeah - I just heard a report of that. Looking in to it now!

Abeefe, 2024-03-17:
OK.

Alien-Tec, 2024-03-24:
Hello.

1. I can confirm, there is no MicroG option in the settings menu.

2. Further, Android Auto does not work with sandboxed Google. There is no option to configure it.
The car says connection failed.

3. BackUp doesn't work because Seedvault is missing.

Hope to see these problems fixed anytime soon.

Regards.

inmaldrerah, 2024-03-26:
I didn't find MicroG in the arm64 microg version (VoltageOS-microg-arm64-ab-3.2-20240316-UNOFFICIAL) either.

Atman-5, 2024-03-28:
There is none in the arm64 vndklite version either, supposing it's missing from all microg builds.

cawilliamson, 2024-04-03:
I've had a nightmare building the newer Voltage sources but am getting on top of it now. This will be addressed in the next build.

closed 2024-04-03 by cawilliamson