cawilliamson/treble_voltage #25: Bug report + suggestion

erickmufath, 2024-03-17:
Respectfully it's up to you, you shouldn't take my notes, but if you don't mind or this suits your preferences and if you can, it would be nice. Bug maybe only on my device-X678B, maybe not. Thank you.

- Some notification not appear in lock screen although i choose to Show All Notification in settings, like progress bar of copying large files in the background (spesifically i use FX File Manager).
- Lockscreen PIN with Face Unlock problem. Swipe to unlock with pin makes the number keys automatically zoom out and and zoom in(return to normal) based on the appearance of the text "Recognizing Face.."
- Cannot switch off SIM signal (This is different from airplane mode).
- Backup settings problem (seedvault missing).
- *#*#4636#*#* missing option to select prefered network frequency or band(not network type like GSM or LTE).
- Cannot switch on battery saver.
- If network permission turn off and turn on again(after boot/reboot) network on that app still not connected to the network until i reboot again.
- Grapheneos Camera problem, when switch from camera(photo) to video the result is like low light, and sometimes error appears.
- Signal icon become double (dual SIM have 4 icon) when swipe down to notification center in lanscape rotation/opening game, it's disappear and normal again only if expanding quick setting in notification center.

- Increase the number of fingerprint inputs for one finger, the stock rom of my device is much more and it makes fingerprints more accurate.
- Add toggle front flashlight.
- Replace Jelly Browser with Vanadium.
- Replace AOSP Keyboard with FlorisBoard Keyboard, AOSP Keyboard have problem like speech to text(mic icon) just a display(don't work) and cannot add or edit keyboard type in setting such a English (US), also it don't have integrated icon.

Atman-5, 2024-03-20:
I'm sure some or most of the issues you described are related to vendor partition of your device and not the system GSI, you need a developer specifically for your device that would want to look into all that.

erickmufath, 2024-03-20:
> I'm sure some or most of the issues you described are related to vendor partition of your device and not the system GSI, you need a developer specifically for your device that would want to look into all that.

Who knows?! Maybe it's from GSI, my partition vendor uses vendor stock and of course there are almost no problems if I use the stock system, is it possible that just changing the stock system to GSI can cause lots of bugs to appear? It would be better to inspect the GSI first. Logically it is possible that other users also experience the same thing and it can be fixed from there.

Alien-Tec, 2024-03-24:
Hi there...

Here are some feature requests from me.

1. Remove AOSP Board and include Openboard + VollaBoard (SpeechToText):
https://fdroid-nightly.volla.tech/fdroid/repo?fingerprint=27c53f6e478e26f751f4fef960ec0640352daa94d0dcf75e1401e89f92863203
https://fdroid-nightly.volla.tech/fdroid/repo/com.volla.vollaboard.apk

2. Give the user the option to remove any preinstalled system app like file manager, gallery and so on - the same way iodéOS does it.
I love the SimpleMobileTools/Fossify - Set and would like to use these apps.

3. Keep on the good work. This GSI is awesome so far.

Regards.

erickmufath, 2024-03-27:
> Hi there...
> 
> Here are some feature requests from me.
> 
> 1. Remove AOSP Board and include Openboard + SayBoard (SpeechToText) + eSpeak (TextToSpeech).
> 2. Give the user the option to remove any preinstalled system app like file manager, gallery and so on - the same way iodéOS does it.
>    I love the SimpleMobileTools/Fossify - Set and would like to use these apps.
> 3. Keep on the good work. This GSI is awesome so far.
> 
> Regards.

Maybe you should create a new issue despite the different requests. I have written about the request to replace the AOSP Keyboard but not SayBoard and about your request number 2.

cawilliamson, 2024-04-03:
Wow, you have clearly taken a lot of time to provide all of this feedback so I'll try to go through it one by one:

**Some notification not appear in lock screen although i choose to Show All Notification in settings, like progress bar of copying large files in the background (spesifically i use FX File Manager).**

Device or ROM issue - not GSI specific.

**Lockscreen PIN with Face Unlock problem. Swipe to unlock with pin makes the number keys automatically zoom out and and zoom in(return to normal) based on the appearance of the text "Recognizing Face.."**

Device or ROM issue - not GSI specific.

**Cannot switch off SIM signal (This is different from airplane mode).**

Backup settings problem (seedvault missing).

★#★#4636#★#★ missing option to select prefered network frequency or band(not network type like GSM or LTE).

Cannot switch on battery saver.

If network permission turn off and turn on again(after boot/reboot) network on that app still not connected to the network until i reboot again.

Grapheneos Camera problem, when switch from camera(photo) to video the result is like low light, and sometimes error appears.

Signal icon become double (dual SIM have 4 icon) when swipe down to notification center in lanscape rotation/opening game, it's disappear and normal again only if expanding quick setting in notification center.

Lag screen when refresh rate adaptive from higher refresh rate to 60hz, feels under 60fps(Youtube Shorts), but if i force to 60hz in phh treble settings, or in display settings set min and max refresh rate to 60hz or same value it feels normal(need reboot to take effect) (More detail : I found 60fps video on Shorts and it look like normal, it made me realize, that this issue occurs only when I touch the screen or anything else that affects changes on the screen except the video being played. Only when scrolling YouTube Shorts video or comments, including when animations or display movements increase or decrease the sound volume with buttons or touchscreen, definetely laggy and feels under 60fps)

Increase the number of fingerprint inputs for one finger, the stock rom of my device is much more and it makes fingerprints more accurate.

Add toggle front flashlight(if detected)

Replace Jelly Browser with Vanadium.

Replace AOSP Keyboard with FlorisBoard Keyboard(Other alternative : OpenBoard), AOSP Keyboard have problem like speech to text(mic icon) just a display(don't work) and cannot add or edit keyboard type in setting such a English (US), also it don't have integrated icon.

19/03/24
New bug reports and suggestions:

I will not demand my report, please take notes if you can and don't mind according to your preferences.

Beside Grapheneos camera problem before, just find out that my 108MP rear and 32MP selfie camera on DevCheck and AIDA64 not detected properly. Rear camera resolution detected just 12M with video resolution just 3.7MP, and selfie camera resolution detected 8.1MP with video resolution also 3.7MP, i think that is the main problem why the camera results are like that.
Result of screen recording a bit blurry, with or without HEVC.
Danmaku Notification Mode in Game Space dissapear too fast(maybe because it's danmaku or because while i playing game?! it feels too fast and sometimes i don't notice the notification). Can you make it more long or make it slowly slide from right to left to make it more noticable?.
Gallery app useless for feature "Unlimited Backup Photos and Videos", makes me still should install other apps like original Google Photos app(maybe remove gallery and let user install whatever user want).
Add charging control from LineagOS, like Voltage 2.8 GSI by ahnet-69
Add toggle to show number of notification in dot notification app setting.
21/03/24

In Airplane Mode with WIFI connected, when expanding quick acces or control center, 2 text "No Service" appear. And when phone locked, there are 2 empty signal icon with 2 same text "No Service". Normaly in Airplane Mode that should be not appear(or it should be "Airplane Mode" text?!).
Monospace font show a bit weird. After installing termux, the font doesn't look good, some alphabet look wider than other. Not monospace font that i know.
Add feature recent apps lock. So app don't close and keep running in the background, even if I press "Clear All" in recent apps, except click app logo in recent apps and unlock it.
22/3/24
Please take notes if this is useful as a reference for making the system better. If necessary, my report can also be forwarded to other developers.

Split screen can only do 50-50 ratio. Other system that i know can do 75-25 or 25-75.
GmsCompat keep show "Sandboxed Google Play is Running" even though i set GmsCompat, Google Service Framework, Google Play Service, and Google play to Restricted in "App battery usage" setting after boot/reboot. I don't know if this is normal and the setting just work like that, just reporting.
Gaming not really smooth in graphic and network(ping) performance stability, maybe can you improve with someting like rendering mode and display compositing?!(Overall, not only on Game Space) Also improving touch response. I am not sure what i am talking about and can or cannot it can be. Also The game ping is too unstable, from 12 ms it can suddenly reach 8ms, 40ms(very unstable), sometimes can reach 200+ms and it can last a long time until the game character can't move. I know that the gaming experience depends on the chipset and network quality, but its grap [... cut ...]
Bluetooth icon doesn't appear in status bar when enabled.
Add setting to enable/disable and control task animation speed like "Ultra Touch" feature in XOS (different from animation scale in developer options which reduces the animation scale, in personal opinion this accelerates the speed of the normal animation scale and/or swipe speed when doing something more than touch response. The effect will be felt on devices with a high refresh rate screen). Of course everything I said earlier applies and this is just an idea.

erickmufath, 2024-04-03:
> Wow, you have clearly taken a lot of time to provide all of this feedback so I'll try to go through it one by one:
> 
> **Some notification not appear in lock screen although i choose to Show All Notification in settings, like progress bar of copying large files in the background (spesifically i use FX File Manager).**
> 
> Device or ROM issue - not GSI specific.
> 
> **Lockscreen PIN with Face Unlock problem. Swipe to unlock with pin makes the number keys automatically zoom out and and zoom in(return to normal) based on the appearance of the text "Recognizing Face.."**
> 
> Device or ROM issue - not GSI specific.
> 
> **Cannot switch off SIM signal (This is different from airplane mode).**
> 
> Device or ROM issue - not GSI specific.
> 
> **Backup settings problem (seedvault missing).**
> 
> Can you provide clarification on this? I was fairly certain Seedvault was present!
> 
> **★#★#4636#★#★ missing option to select prefered network frequency or band(not network type like GSM or LTE).**
> 
> Device or ROM issue - not GSI specific.
> 
> **Cannot switch on battery saver.**
> 
> Can you provide additional information here?
> 
> **If network permission turn off and turn on again(after boot/reboot) network on that app still not connected to the network until i reboot again.**
> 
> That is strange but is very likely a ROM issue since there is nothing GSI specific here. May be worth raising this with the VoltageOS developers.
> 
> **Grapheneos Camera problem, when switch from camera(photo) to video the result is like low light, and sometimes error appears.**
> 
> I have removed GrapheneOS camera and reverted back to the LineageOS Aperture camera app in the new builds (coming soon - just working out the bugs!)
> 
> **Signal icon become double (dual SIM have 4 icon) when swipe down to notification center in lanscape rotation/opening game, it's disappear and normal again only if expanding quick setting in notification center.**
> 
> Device or ROM issue - not GSI specific.
> 
> **Lag screen when refresh rate adaptive from higher refresh rate to 60hz, feels under 60fps(Youtube Shorts), but if i force to 60hz in phh treble settings, or in display settings set min and max refresh rate to 60hz or same value it feels normal(need reboot to take effect) (More detail : I found 60fps video on Shorts and it look like normal, it made me realize, that this issue occurs only when I touch the screen or anything else that affects changes on the screen except the video being played. Only when scrolling YouTube Shorts video or comments, including when animations or display movements increase or decrease the sound volume with buttons or touchscreen, definetely laggy and feels under 60fps)**
> 
> Device or ROM issue - not GSI specific.
> 
> **Increase the number of fingerprint inputs for one finger, the stock rom of my device is much more and it makes fingerprints more accurate.**
> 
> ROM request - should be directed to VoltageOS development team.
> 
> **Add toggle front flashlight(if detected)**
> 
> ROM request - should be directed to VoltageOS development team.
> 
> **Replace Jelly Browser with Vanadium.**
> 
> I will consider this - Jelly Browser is, indeed, not my favourite by far!
> 
> **Replace AOSP Keyboard with FlorisBoard Keyboard(Other alternative : OpenBoard), AOSP Keyboard have problem like speech to text(mic icon) just a display(don't work) and cannot add or edit keyboard type in setting such a English (US), also it don't have integrated icon.**
> 
> **Beside Grapheneos camera problem before, just find out that my 108MP rear and 32MP selfie camera on DevCheck and AIDA64 not detected properly. Rear camera resolution detected just 12M with video resolution just 3.7MP, and selfie camera resolution detected 8.1MP with video resolution also 3.7MP, i think that is the main problem why the camera results are like that.**
> 
> Device or ROM issue - not GSI specific.
> 
> May be resolved by switch back to Aperture in coming builds.
> 
> **Result of screen recording a bit blurry, with or without HEVC.**
> 
> Device or ROM issue - not GSI specific.
> 
> **Danmaku Notification Mode in Game Space dissapear too fast(maybe because it's danmaku or because while i playing game?! it feels too fast and sometimes i don't notice the notification). Can you make it more long or make it slowly slide from right to left to make it more noticable?**
> 
> ROM request - should be directed to VoltageOS development team.
> 
[... 73 lines cut ...]
> 
> Device or ROM issue - not GSI specific.
> 
> **Bluetooth icon doesn't appear in status bar when enabled.**
> 
> Device or ROM issue - not GSI specific.
> 
> **Add setting to enable/disable and control task animation speed like "Ultra Touch" feature in XOS (different from animation scale in developer options which reduces the animation scale, in personal opinion this accelerates the speed of the normal animation scale and/or swipe speed when doing something more than touch response. The effect will be felt on devices with a high refresh rate screen). Of course everything I said earlier applies and this is just an idea.**
> 
> ROM request - should be directed to VoltageOS development team.
> 
> **Summary:**
> 
> A lot of what you're asking for is either:
> 
> 1. Feature requests - for the most part these belong with the VoltageOS team - I don't build the ROM - I just adapt it for GSIs.
[... cut ...]  
- Cannot enable battery saver. In the control center it cannot be activated at all, you can click it but nothing happens, even though a low battery notification appears and you ask to activate it from the notification it also doesn't change anything. 

Also, are you really unable to adjust the rendering mode or anything regarding the graphics? something like skiagl or vulkan. I've seen it while browsing GitHub which seems to contain several graphic settings and others with the name system.prop

erickmufath, 2024-04-03:
[image]
[image]

cawilliamson, 2024-04-28:
Pretty much everything you're describing there are issues with either device specific patches or Voltage itself - I'm just adapting Voltage to run as a GSI - I can't adapt / fix / patch every single problem with it unfortunately. I just don't have the resources and it would essentially be a different ROM at that point.

For device specific stuff I suggest raising a ticket at https://github.com/TrebleDroid/treble_experimentations/issues although I would try to keep your ticket more short and to the point or it'll likely be ignored there. :)

For Voltage stuff... don't really know what to suggest. I found them to be incredibly hostile when I suggested a change so I don't communicate with them at all.

closed 2024-04-28 by cawilliamson