cawilliamson/treble_voltage #28: Feature request

Alien-Tec, 2024-03-27:
As written here
#https://github.com/cawilliamson/treble_voltage/issues/25#issuecomment-2016825217

Hi there...

Here are some feature requests from me.

    Remove AOSP Board and include Openboard + SayBoard (SpeechToText) + eSpeak (TextToSpeech).

    Give the user the option to remove any preinstalled system app like file manager, gallery and so on - the same way iodéOS does it.
    I love the SimpleMobileTools/Fossify - Set and would like to use these apps.

    Keep on the good work. This GSI is awesome so far.

Regards.

cawilliamson, 2024-04-03:
**Remove AOSP Board and include Openboard + SayBoard (SpeechToText) + eSpeak (TextToSpeech).**

Removing and replacing packages shipped with Voltage has proven to be problematic due to deep integrations with various services so I'm unlikely to implement this.

**Give the user the option to remove any preinstalled system app like file manager, gallery and so on - the same way iodéOS does it. I love the SimpleMobileTools/Fossify - Set and would like to use these apps.**

I do completely understand the desire here - the problem is if every app is uninstallable some users will naturally break things. Although I believe all applications that you would want to disable can be, right? No?

Alien-Tec, 2024-04-03:
I guess you are right. 👍🏻
Root the phone, uninstall everything that's not needed and install the New apps.

closed 2024-04-28 by cawilliamson