cawilliamson/treble_voltage #30: face unlock not working

chetan-go, 2024-04-08:
for me all other things are working except face unlock other than that this gsi is wow i am using redmi 9i and there is even brightness slider bug it does not work properly for me i dont know if it is gsi or device issue

erickmufath, 2024-04-09:
I have no issue about face unlock and brightness slider, so it's possibly a device issue

chetan-go, 2024-04-10:
maybe

chetan-go, 2024-04-10:
what about new ver will we get any update

cawilliamson, 2024-04-28:
It's 100% a device issue - you should raise a ticket on https://github.com/TrebleDroid/treble_experimentations/issues for your specific device and add as much info about your device as possible.

If your decice is arm32_binder64 though I wouldn't expect this to work.

New update coming as soon as I can build it and get it to boot!

closed 2024-04-28 by cawilliamson