cawilliamson/treble_voltage #37: Microg build still doesn't seem to work

erickmufath, 2024-05-15:
[image]
[image]
1. Logically Apps by grapheneos that can help install gapps should be gone because their job has been replaced by MicroG, but it's still there.
2. Because MicroG apps and settings missing or still wasn't working, i tried to install gapps via Apps by grapheneos, but the result is error, i think some of the MicroG configurations that were "running" conflicted with it.

Alien-Tec, 2024-05-15:
@cawilliamson 
Please, please just get rid oft gapps.
There are so many custom roms out there that offer gapps build only.
Simply stick to vanilla and microG.
That would be awesome.

erickmufath, 2024-05-18:
> > @cawilliamson Please, please just get rid of gapps. There are so many custom roms out there that offer gapps builds only. Simply stick to vanilla and microG. That would be awesome.
> 
> gapps isn't causing the issue - GmsCompat is. I'm working on it and only adding GmsCompat to vanilla builds.

I never install Voltage with Gapps build, but same as MicroG build, if Apps by grapheneos is still there, i think it's not necessary. Also all configuration about that app. Currently used Vanilla build before MicroG works.

cawilliamson, 2024-05-20:
Fixed in latest release - have fun! :)

closed 2024-05-20 by cawilliamson