cawilliamson/treble_voltage #43: [bug？]Screen goes black

tokumei123456, 2024-08-23:
Symptoms: The screen goes black.

〈When does this happen?〉
It happens irregularly.
But it happens in two ways:
・When switching from an app to the home screen
・When opening an app from the home screen

〈Solution〉
Press the “task management button” or “home button” several times. (These are the ■ and ● buttons.)

〈Inexplicable points〉
・The status bar is displayed.
・The buttons at the bottom of the screen (◀, ●, ■) are displayed.
・The quick settings screen can be opened.

Usage environment
・Xperia 5 IV
・Rooted
・VoltageOS-vanilla-arm64-ab-3.5-20240731-UNOFFICIAL.img.xz

cawilliamson, 2024-08-27:
I've just pushed a new build - very few TD changes but there are some VoltageOS changes - hopefully that helped! :+1: 

If not - this will likely need to be raised with the fine folks over at https://github.com/TrebleDroid/treble_experimentations/issues since it sounds like a TrebleDroid patches issue instead of a ROM issue.

There's nothing very GSI specific in the ROM which would cause this.

closed 2024-08-27 by cawilliamson

tappmj, 2024-09-08:
I also encountered a bug with VoltageOS-vanilla-arm64-ab-3.6-20240827-UNOFFICIAL.img.xz that caused some screens to turn black, so I will report this to https://github.com/TrebleDroid/treble_experimentations/issues.