cawilliamson/treble_voltage #52: Play Integrity failure.

GamePlayer-8, 2024-11-28:
Good evening,

I've stumbled upon your project about providing the GSI images of VoltageOS and I'm thankful to you for making them available to everyone.

However, whenever I'm trying to bypass Play Integrity check, with [Shamiko](https://github.com/LSPosed/LSPosed.github.io/releases), [PlayIntegrityFix](https://github.com/chiteroman/PlayIntegrityFix/releases) and [ZygiskNext](https://github.com/Dr-TSNG/ZygiskNext/releases) accompanied by the [Magisk project](https://github.com/topjohnwu/Magisk/releases), I wasn't been able to make SafetyNet as well as Play Integrity work anyhow.

Once I install root onto the image, even the Basic Integrity starts failing without any way to fix.

Whenever I'm trying to flash my custom GPlay onto the image, is being rejected by the image.

I'm reporting it as a bug as of it wasn't happening in 2.7 (I know, old version).

iabeefe, 2024-12-31:
Try ths

https://xdaforums.com/t/achieve-strong-integrity-with-working-rcs-all-oem.4697473/

cawilliamson, 2025-03-01:
Play Integrity does not pass on GrapheneOS which VoltageOS is based on.

This is not a GSI issue.

closed 2025-03-01 by cawilliamson