cawilliamson/treble_voltage #58: wrong SMSC  number / I can't receive or send text messages on samsung

morosakubek2, 2025-03-12:
Hi,
Hi I have a problem on my samsung phone with sending and receiving text messages. SMSC shows the number: 48602951111 and it should be with quotes " 48602951111",145. Updating does not change the settings.

best,
Morosa

morosakubek2, 2025-03-12:
Update: The smsc update works, but only digits can be inserted.

cawilliamson, 2025-03-13:
Oh weird! Sounds like a device specific issue. I'd suggest raising that https://github.com/TrebleDroid/treble_experimentations which this ROM is using patches based on.

morosakubek2, 2025-03-18:
Hi, thanks for your reply. Yes, it looks like it applies to all new samsungs. Maybe this would solve the problem: https://github.com/LineageOS/android_device_samsung_a71-common/commit/33545c7b467a3671f194c360e9496bf289364e77