cawilliamson/treble_voltage #61: Bluetooth Bug

Technical06, 2025-03-17:
also a part of Evox by mytja A14 fixes

cawilliamson, 2025-03-18:
Android 14 is not Android 15. Fixes aren't portable but nobody else has reported any actual bluetooth issues on this firmware so I suspect they are device specific and not related to the ROM. Be sure to turn on the relevant workarounds under Bluetooth Workarounds in TrebleApp.

iabeefe, 2025-03-19:
@cawilliamson The bleutooth bug for me is that without sound from connected device, bluetooth disconnect and reconnect consistently.

Technical06, 2025-03-21:
> I have all of the BT fixes from mytja at this point.
> 
> Have you gone into the TrebleApp settings and enabled the MediaTek bluetooth workarounds?

yeah i tried that , and selected predefined workarounds to mediatek , still didnt worked

Technical06, 2025-03-21:
> I have all of the BT fixes from mytja at this point.
> 
> Have you gone into the TrebleApp settings and enabled the MediaTek bluetooth workarounds?

Bluetooth bug is still not fixed

cawilliamson, 2025-03-21:
The latest build rolls back the patch I just applied which seemingly made things worse but does not include any new fixes.

I'm confused though because I run a MTK device and have zero issues.

Can people tell me what devices they're experiencing this on and you mentioned there's patches for it - do you know which ones? A15-QRP1 changed a lot of stuff (Again) and broke a lot of stuff (again, thanks Google!)

cawilliamson, 2025-03-21:
Also - has anyone here tried the A15 EvoX builds from mytja and if so - do they work properly with Bluetooth?

cawilliamson, 2025-03-21:
```
getprop persist.sys.bt.unsupported.commands
```

Can someone run this on their device who has broken BT *after* you have triggered the disconnect/reconnect bug?

iabeefe, 2025-03-23:
[image]

> ```
> getprop persist.sys.bt.unsupported.commands
> ```
> 
> Can someone run this on their device who has broken BT _after_ you have triggered the disconnect/reconnect bug?

iabeefe, 2025-03-23:
[log with device info.zip](https://github.com/user-attachments/files/19409666/log.with.device.info.zip)

Technical06, 2025-03-24:
> ```
> getprop persist.sys.bt.unsupported.commands
> ```
> 
> Can someone run this on their device who has broken BT _after_ you have triggered the disconnect/reconnect bug?

ill check both mytja A15 build and this command

Technical06, 2025-03-24:
> ```
> getprop persist.sys.bt.unsupported.commands
> ```
> 
> Can someone run this on their device who has broken BT _after_ you have triggered the disconnect/reconnect bug?

[image]
he/she has mentioned a bluetooth fix , lets test it

Technical06, 2025-03-24:
> Also - has anyone here tried the A15 EvoX builds from mytja and if so - do they work properly with Bluetooth?

i tried it , it has the same issue as voltage OS and probably other gsi ( I dont know about ) ill try running the command u mentioned , do i need to run it in terminal or platform tools?

cawilliamson, 2025-04-12:
Could you try the new Bluetooth option added to TrebleApp and see if that helps at all in the latest build?

closed 2025-05-14 by cawilliamson