cawilliamson/treble_voltage #62: Instruction/help

RogerioTDS, 2025-03-20:
Hello.
First of all, thank you for making the ROM available.
I'm having trouble opening two banking apps. Do you have any suggestions on how to get around this?
I'm using the latest version of gapps, or more precisely, this one:

https://github.com/cawilliamson/treble_voltage/releases/download/4.2-20250313.120122/VoltageOS-gapps-arm64-ab-4.2-20250313.105743-UNOFFICIAL.img.xz

cawilliamson, 2025-03-20:
For advice on hiding root I'd chat in: https://t.me/revenychat
For advice on using the ROM I'd chat in: https://t.me/ahnetgsi

I'll close this ticket though since it's not something I can help with personally I'm afraid!

closed 2025-03-20 by cawilliamson

RogerioTDS, 2025-03-20:
Hello, thank you for your attention.
The only way to do this is using root, right?
I don't intend to use root at the moment.
If the ROM "hide" the security update date, like Ponces does, I think it would work.
Because Ponces's ROM "hides" this information, so the apps work there. For example: instead of showing Android Security Update February 5th, the security date of the stock ROM appears, and they work.
I think I'll have to go back to it.
Thank you in advance for your attention.