cawilliamson/treble_voltage #64: Google Play Certification Error in latest build

Technical06, 2025-03-20:


cawilliamson, 2025-03-20:
That's not really a target of the GSI so I'm afraid if it's broken - it will likely remain so.

People on custom ROMs are almost universally suffering this same issue. It's not really worth trying to get play certification anymore.

Technical06, 2025-03-20:
> That's not really a target of the GSI so I'm afraid if it's broken - it will likely remain so.
> 
> People on custom ROMs are almost universally suffering this same issue. It's not really worth trying to get play certification anymore.

its showing error this device is not play protect certified

cawilliamson, 2025-03-21:
I believe the problem you were seeing was related to a conflict between the ponces "securize" patch which spoofs lots of props and the built-in GMS device spoofing in Voltage.

I've added a change which should ONLY apply the spoofed props from "securize" if you're not running a full gapps build. Please try the latest build and report back. :)

https://github.com/cawilliamson/treble_voltage/releases/tag/4.2-20250321.142314

iabeefe, 2025-03-23:
> I believe the problem you were seeing was related to a conflict between the ponces "securize" patch which spoofs lots of props and the built-in GMS device spoofing in Voltage.
>
> I've added a change [...]

I have device integrity with this build. Thanks.

closed 2025-03-23 by cawilliamson

Technical06, 2025-03-24:
> Awesome, thanks for confirming! I'll close this issue.

i tried the latest build but the device not certified error is still not fixed also the GSF id shows empty so i can't even register it as an uncertified device

Technical06, 2025-03-24:
> Awesome, thanks for confirming! I'll close this issue.

there's no gsf id in the latest build