cawilliamson/treble_voltage #65: no fingerprint VoltageOS 4.2-20250321.142314

matheuszig, 2025-03-24:
device galaxy a11 - arm32 bind 64 vanilla. Fingerprint not work and not show any options to lockscreen with fingerprint.

Technical06, 2025-03-24:
Well about this issue , A15 build of mytja has fixed fingerprint issue for my device , idk about other devices but fingerprint is working fine in that A15 build of Evox 10.3 , @cawilliamson do u mind putting those patches in voltage OS , as for the bluetooth in A15 evox it still exists ,

matheuszig, 2025-03-24:
> Well about this issue , A15 build of mytja has fixed fingerprint issue for my device , idk about other devices but fingerprint is working fine in that A15 build of Evox 10.3 , [...]

your device is samsung ?

cawilliamson, 2025-03-25:
This is a duplicate of issue #54 - please don't clutter up the tickets with duplicates.

closed 2025-03-25 by cawilliamson