cawilliamson/treble_voltage #66: After Installing latest version of voltage OS , device is uncertified + GSF id not detected by device id tool , so i can't even register it manually

Technical06, 2025-03-26:
if u find thisi duplicate , you may close it , I created this issue as there was no response , maybe you're on a break after working so hard

cawilliamson, 2025-04-14:
This only happens on Play Services uncertified devices (i.e. devices which are not certified with Google) *but* I have a solution in the next build.

By default now Securize is on (this will break play certification for your devices) *BUT* the TrebleApp now has an option to disable it which will turn off all prop spoofing related to certification. This means your device will identity as "TrebleDroid GSI" or similar *but* it will be Google certified.

I will get this release out shortly - keen an eye out. 👍

cawilliamson, 2025-04-14:
https://github.com/cawilliamson/treble_voltage/releases/tag/4.2-20250414.181137

This release solves the problem - if your device is uncertified for play services go in to TrebleApp and disable "Enable Securize", then reboot. This will resolve your problem.

closed 2025-04-14 by cawilliamson