cawilliamson/treble_voltage #72: Launcher bug.

VasilKotsov, 2025-04-22:
Hello I try your rom and i really like it. But I have some kind of bug with the launcher its stuck on black screen and crash it. 

[Error in Voltage Home 84a526df3058.txt](https://github.com/user-attachments/files/19848096/Error.in.Voltage.Home.84a526df3058.txt)

cawilliamson, 2025-05-14:
Has this improved with the newer builds?

VasilKotsov, 2025-05-14:
I don't know because I using a LineageOS GSI form [MisterZtr](https://github.com/MisterZtr). The battery life is a bit better i charged my device at Sunday and now i charging again but with your ROM the battery is discharged through the night with around 20%. But I will try your ROM again to see if still this bug with the launcher appear.