cawilliamson/treble_voltage #73: Bluetooth crash

Ridhoardhiansyah7, 2025-04-23:
I use a Bluetooth headset. When I use the headset while connected to Wi-Fi, the Bluetooth connection often crashes. However, when using mobile data, the Bluetooth works fine.

 Here’s the crash log :

[Error in Bluetooth 6530002ab2c0.txt](https://github.com/user-attachments/files/19860678/Error.in.Bluetooth.6530002ab2c0.txt)

cawilliamson, 2025-05-14:
Please try the new toggle related to this in TrenleApp.

closed 2025-05-19 by cawilliamson

Ridhoardhiansyah7, 2025-05-23:
Ok I will try it later, my device realme narzo 50i/c11 2021,  i'm using gsi with arch arm32-binder64

the device has 2 bugs that occur in all gsi roms, backlight (can be fixed) and cannot use a wired headset (this can be fixed in gsi rom android version <= 12), to fix the backlight bug in gsi rom, my device requires root access or Rooted Debugging in developer option

This gsi cannot be rooted with magisk and does not have the 'Rooted Debugging' I cannot fix the backlight bug mentioned, can you add a patch or config for the phone? Thank you

The backlight bug can be fixed with this command: setprop persist.sys.qcom-brightness 4500

Sorry my english is very bad