cawilliamson/treble_voltage #76: Bootloop

root-Dante42, 2025-05-09:
On last build VoltageOS 4.3-20250507.222825 my phone (moto g62) go in boot loop

root-Dante42, 2025-05-11:
same on build VoltageOS 4.3-20250509.230912

iabeefe, 2025-05-12:
Having the same issue here. Stuck at boot animation.

cawilliamson, 2025-05-14:
I know this is tricky but without any logging at all this is going to be extremely difficult to track down the problem. Is there no adb logcat output at all?

iabeefe, 2025-05-17:
@cawilliamson  How to get logs? And the boot animation is stuck at the bottom right of the screen. I use a 720 screen.

root-Dante42, 2025-05-18:
@cawilliamson  I tried to do an adb logcat but the phone doesn't even reach the adb

cawilliamson, 2025-05-19:
Try the latest experimental build and report back :)

root-Dante42, 2025-05-19:
same problem on experimental build

cawilliamson, 2025-05-26:
Are you able to run all other up-to-date GSIs on A15-QRP2?

root-Dante42, 2025-05-26:
up to build VoltageOS-vanilla-arm64-ab-4.2-20250414 all builds work, all builds after that don't

iabeefe, 2025-05-26:
No! My device is a Tecno MTK a64.

iabeefe, 2025-05-26:
OK.

iabeefe, 2025-05-27:
@cawilliamson I can test it if built.

It is not that  the OS is not booting, it is just stuck at boot animation and the voltage OS is stuck at the bottom right corner of the screen

root-Dante42, 2025-05-27:
last build boot and work

cawilliamson, 2025-05-27:
Try to set the prop to 0 via adb, if you can't get to it you can downgrade to an older image which works for you and then inset it, then flash the newer image again

If all else fails you'll have to wipe your data.

root-Dante42, 2025-05-27:
how can i set to 0

root-Dante42, 2025-05-27:
i can't lost data on phone

root-Dante42, 2025-05-28:
I solved it like this I first tried to disable the option then it didn't work I installed the latest release of infinity x then disabling the option "Treat virtual sensors as real" and finally installing the latest voltage but the wifi network and the 4g network don't work

cawilliamson, 2025-05-28:
@iabeefe Did the new build do anything new on your device? :)