cawilliamson/treble_voltage #79: Bootloop on the Doogee S110

ipjustice, 2025-05-12:
I performed a flash using ADB, but the device fails to reach the startup animation before restarting.

cawilliamson, 2025-05-26:
Please use https://github.com/cawilliamson/treble_voltage/issues/76 to further progress this issue 👍

ipjustice, 2025-05-26:
Well, I've used the latest Vanilla Arm64-ab one and these are the steps I took in flashing it

[Steps.txt](https://github.com/user-attachments/files/20444472/Steps.txt)

After doing that I rebooted, then the phone hanged on the logo for a minute then rebooted itself,
It booted to a screen complaining about data corruption with an option to factory reset it, I chose to factory reset it, and it rebooted and hanged on the logo again and rebooted itself to the data corruption screen again
And the loop stayed like this.

cawilliamson, 2025-05-26:
Thanks for the info - are you able to run any other TrebleDroid GSIs based on Android 15-QPR2 at all? Wondering if this is an issue with a "personal" patch or a TrebleDroid patch!

cawilliamson, 2025-05-26:
Whilst in the "your data may be corrupt" screen could you try running the following actually?

```
adb shell cat /proc/last_kmsg
```

Then find a way to copy that output and paste it on this ticket or attach it as a file? That would be incredibly helpful to see what may be going wrong here.

ipjustice, 2025-05-26:
I'm going to try to install stock then authorize and flash the GSI after, maybe that'll work

cawilliamson, 2025-05-26:
Ahhh, if you can't run any GSIs then you may be better off posting on:

https://github.com/TrebleDroid/treble_experimentations/issues

I wasn't aware that NO GSIs were working on your device at all. That's going to be very difficult for me to try to resolve from my end unfortunately!

ipjustice, 2025-05-26:
I didn't mean that ALL of them don't work what I meant to say that some specific ones only work like this one:
https://sourceforge.net/projects/andyyan-gsi/files/lineage-20-td/
This could be a hint on the issue.
And regarding commenting on the treble_expermentations thread, I believe I won't get a response for the next 6 months lol, but I'll try, thanks!

ipjustice, 2025-05-26:
That lineage is flawless, the only issue is smoothness

cawilliamson, 2025-05-27:
5.10 isn't so old that you should be having problems indeed. That's strange!

I'm compiling a build right now which contains a few experimental fixes - with any luck one of those fixes your device - then it's just a case of tracking down which one!

ipjustice, 2025-05-27:
I've tried multiple things and this is what I've came up with:

This one also fails sadly.(With & without DSU)
I've tried using the one this project was forked from to no avail too.

But get this: When I use DSU sideloader with [this build](https://github.com/cawilliamson/treble_voltage/releases/tag/4.2-20250313.120122) the phone reaches the boot animation but gets stuck there, flashing this one directly still results in the phone not booting but might give another clue.
I'll try to use the latest 4.2 build to see.

cawilliamson, 2025-05-29:
Ok I'll make you both a debug build so you get logcat access. Let's try to get this sorted!

iabeefe, 2025-05-30:
> https://github.com/cawilliamson/treble_voltage/releases/tag/4.3-20250529.070616
> 
> Please test this build - it should allow unauthenticated logcat output. Provide as much logging as possible and I will investigate the bootloop issues. :)

What is the adb command to run to get logs?

ipjustice, 2025-05-30:
@iabeefe adb logcat -all

cawilliamson, 2025-05-31:
If I'm not mistaken it should be `adb logcat -b all`, no? Maybe -all is a shortcut. Not sure. 👍

iabeefe, 2025-05-31:
OK.

Proof-12, 2025-06-01:
My samsung a13 32bit exynox boot your 4.2 to 4.3 version of your voltage os gsi (experimental also but not debug release)i have kernel 4.19

iabeefe, 2025-06-01:
[image]

I can not get the logcat.

ipjustice, 2025-06-02:
Someone has managed to get the logcat.
(He has the same MTK chipset and device)

2025-05-30 17:35:02.931 1003-1003 hwcomposer pid-1003 E [IPqDevice] Can't get PQ service tried (0) times
2025-05-30 17:35:02.932 335-4244 libc pid-335 W Unable to set property "ctl.interface_start" to "vendor.mediatek.hardware.pq@2.14::IPictureQuality/default": error code: 0x20
2025-05-30 17:35:02.932 335-4244 hwservicemanager pid-335 I Tried to start vendor.mediatek.hardware.pq@2.14::IPictureQuality/default as a lazy service, but was unable to. Usually this happens when a service is not installed, but if the service is intended to be used as a lazy service, then it may be configured incorrectly.
2025-05-30 17:35:02.948 335-335 hwservicemanager pid-335 I Since vendor.mediatek.hardware.pq@2.14::IPictureQuality/default is not registered, trying to start it as a lazy HAL.
2025-05-30 17:35:02.948 1003-1003 hwcomposer pid-1003 E [IPqDevice] Can't get PQ service tried (0) times

ipjustice, 2025-06-02:
> Someone has managed to get the logcat.
> (He has the same MTK chipset and device)
>
>
> 2025-05-30 17:35:02.931 1003-1003 hwcomposer pid-1003 E [IPqDevice] Can't get PQ service tried (0) times
> 2025-05-30 [...]

@zerofrip