cawilliamson/treble_voltage #82: Home launcher crash

mejiblets, 2025-05-19:
Home launcher crashes on all 4.3 versions when using 3 button navigation bar. When 3 BNB is used it flickers constantly then becomes usable oafter a few minutes but can`t go to home screen otherwise home crashes. Using gestures instead is the workaround but personally I prefer 3BNB

cawilliamson, 2025-05-19:
There's another ticket for this - please update that with logs of you want this investigating. :)

closed 2025-05-19 by cawilliamson

mejiblets, 2025-05-26:
Hi, I've attached the error log for the home screen crashing when using 3 button navigation bar. If it helps, the boot animation also doesn't fit in the centre of the screen when booting, possibly an issue with screen size and given the screen size is also shown in the log.

Is there a way to manually adjust the screen dimensions and I'll try it and see if it makes the 3 button navigation bar work properly.

Thanks

type: logcat
osVersion: samsung/gta7litewifixx/gta7litewifi:12/SP1A.210812.016/T220XXS9EYA4:user/release-keys
flags: dev options enabled
package: com.android.launcher3:35, targetSdk 35
buffers: main,system,crash,events,kernel
level: verbose

--------- beginning of main
05-26 08:33:31.343  2457  2457 I libc    : SetHeapTaggingLevel: tag level set to 0
05-26 08:33:31.373  2457  2457 I droid.launcher3: Using CollectorTypeCC GC.
05-26 08:33:31.386  2457  2457 W droid.launcher3: Core platform API violation: Ljava/lang/reflect/Field;->accessFlags:I from Landroid/os/Build; using reflection
--------- beginning of kernel
05-26 08:33:31.394  2457  2457 I (6)[2457:droid.launcher3]binder: 2457:2457 ioctl 40046210 7ffe53f054 returned -22
--------- switch to main
05-26 08:33:31.516  2457  2457 I droid.launcher3: AssetManager2(0x7483e7c718) locale list changing from [] to [en-GB]
05-26 08:33:31.552  2457  2457 I QuickstepProtoLogGroup: Initializing ProtoLog.
05-26 08:33:31.591  2457  2457 I droid.launcher3: AssetManager2(0x7483e7ae18) locale list changing from [] to [en-GB]
05-26 08:33:31.595  2457  2457 I droid.launcher3: AssetManager2(0x7483e77f38) locale list changing from [] to [en-GB]
05-26 08:33:31.608  2457  2457 I droid.launcher3: AssetManager2(0x7483e7c3f8) locale list changing from [] to [en-GB]
05-26 08:33:31.619  2457  2457 I DisplayController: (CTOR) perDisplayBounds: {CachedDisplayInfo{size=Point(800, 1340), cutout=DisplayCutout{insets=Rect(0, 0 - 0, 0) waterfall=Insets{left=0, top=0, right=0, bottom=0} boundingRect={Bounds=[Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0)]} cutoutPathParserInfo={CutoutPathParserInfo{displayWidth=0 displayHeight=0 physicalDisplayWidth=0 physicalDisplayHeight=0 density={0.0} cutoutSpec={} rotation={0} scale={0.0} physicalPixelDisplaySizeRatio={0.0}}} sideOverrides=null}, rotation=0}=[WindowBounds{bounds=Rect(0, 0 - 800, 1340), insets=Rect(0, 37 - 0, 0), availableSize=Point(800, 1303), rotationHint=0}, WindowBounds{bounds=Rect(0, 0 - 1340, 800), insets=Rect(0, 37 - 0, 0), availableSize=Point(1340, 763), rotationHint=1}, WindowBounds{bounds=Rect(0, 0 - 800, 1340), insets=Rect(0, 37 - 0, 0), availableSize=Point(800, 1303), rotationHint=2}, WindowBounds{bounds=Rect(0, 0 - 1340, 800), insets=Rect(0, 37 - 0, 0), availableSize=Point(1340, 763), rotationHint=3}]}
05-26 08:33:33.281  2457  2457 I ContextualSearchStMgr: Sent LPH override to sysui: 0;0.0
05-26 08:33:34.274  2457  2457 I droid.launcher3: AssetManager2(0x7483e79518) locale list changing from [] to [en-GB]
05-26 08:33:34.275  2457  2457 I droid.launcher3: AssetManager2(0x7483e75378) locale list changing from [] to [en-GB]
05-26 08:33:34.277  2457  2457 I droid.launcher3: AssetManager2(0x7483e79b58) locale list changing from [] to [en-GB]
--------- beginning of system
05-26 08:33:34.282  2457  2457 I ActivityThread: ApplicationInfo updating for com.android.launcher3, new timestamp: 86292
05-26 08:33:34.282  2457  2457 I ActivityThread: assets removed: [/data/resource-cache/com.android.systemui-accent-7xrD.frro, /data/resource-cache/com.android.systemui-dynamic-ad99.frro, /data/resource-cache/com.android.systemui-neutral-f2Ol.frro]
05-26 08:33:34.282  2457  2457 I ActivityThread: assets added: [/data/resource-cache/com.android.systemui-neutral-gKms.frro, /data/resource-cache/com.android.systemui-accent-pMlk.frro, /data/resource-cache/com.android.systemui-dynamic-CgvK.frro]
--------- switch to main
05-26 08:33:34.338  2457  2457 I droid.launcher3: AssetManager2(0x7483e74a18) locale list changing from [] to [en-GB]
05-26 08:33:34.343  2457  2457 I droid.launcher3: AssetManager2(0x7483e714f8) locale list changing from [] to [en-GB]
05-26 08:33:34.379  2457  2457 I droid.launcher3: AssetManager2(0x7483e73758) locale list changing from [] to [en-GB]
05-26 08:33:34.381  2457  2457 I droid.launcher3: AssetManager2(0x7483e746f8) locale list changing from [] to [en-GB]
05-26 08:33:34.382  2457  2457 I droid.launcher3: AssetManager2(0x7483e6f5b8) locale list changing from [] to [en-GB]
--------- switch to system
05-26 08:33:34.383  2457  2457 I ActivityThread: ApplicationInfo updating for com.android.launcher3, new timestamp: 86390
05-26 08:33:34.383  2457  2457 I ActivityThread: assets removed: []
05-26 08:33:34.383  2457  2457 I ActivityThread: assets added: []
--------- switch to main
05-26 08:33:34.389  2457  2457 I droid.launcher3: AssetManager2(0x7483e6f8d8) locale list changing from [] to [en-GB]
05-26 08:33:34.392  2457  2457 I droid.launcher3: AssetManager2(0x7483e78ed8) locale list changing from [] to [en-GB]
05-26 08:33:35.976  2457  2457 I OverviewComponentObserver: updateOverviewTargets: mIsHomeDisabled=false, isDefaultHomeNull=false, mIsDefaultHome=true
05-26 08:33:36.147  2457  2457 I droid.launcher3: AssetManager2(0x7483e775d8) locale list changing from [] to [en-GB]
05-26 08:33:36.154  2457  2457 I droid.launcher3: AssetManager2(0x7483e759b8) locale list changing from [] to [en-GB]
--------- switch to kernel
[... cut ...]
05-26 08:36:37.418  4228  4228 I droid.launcher3: AssetManager2(0x7483e6e2f8) locale list changing from [] to [en-GB]
05-26 08:36:37.424  4228  4228 I droid.launcher3: AssetManager2(0x7483e7c3f8) locale list changing from [] to [en-GB]
05-26 08:36:37.443  4228  4228 I droid.launcher3: AssetManager2(0x7483e775d8) locale list changing from [] to [en-GB]
05-26 08:36:37.456  4228  4228 I DisplayController: (CTOR) perDisplayBounds: {CachedDisplayInfo{size=Point(800, 1340), cutout=DisplayCutout{insets=Rect(0, 0 - 0, 0) waterfall=Insets{left=0, top=0, right=0, bottom=0} boundingRect={Bounds=[Rect(0, 0 - 0, 0), R